fischermoseley/manta #14: Yosys-compatible verilog

TheZoq2, 2023-10-28:
This seems like a really cool project which I'm excited to use, but unfortunately, it doesn't seem to work very well with yosys, at least targeting ECP5s.

First, there are some compilation errors in the block_memory module

The type casts here aren't supported
```verilog
    localparam N_BRAMS = int'($ceil(real'(WIDTH) / 16.0));
```
and neither are these multi-dim arrays.
```verilog
    reg [N_BRAMS-1:0][ADDR_WIDTH-1:0] addra = 0;
    reg [N_BRAMS-1:0][15:0] dina = 0;
```
There may be others, but I looked to other methods at this point, in particular running through https://github.com/zachjs/sv2v takes care of most of these, but the `real` cast is still a problem.

With sv2v and some manual intervention in the `N_BRAMS` computation I was able to get it to compile, but yosys seems to fail to map the BRAM to BRAMs. I can't see an obvious reason for that :/

fischermoseley, 2024-03-07:
Oops - it seems like GitHub auto-closed this when I wasn't looking for it to.

So sorry for the late reply on this! I rewrote the entire codebase in Amaranth HDL, and those tweaks also include a fix for this problem. Go ahead and give the latest release a try if you're still interested :)

fischermoseley, 2024-04-15:
Closing due to inactivity - feel free to reopen if you run into troubles!

closed 2024-04-15 by fischermoseley