fischermoseley/manta #20: help with 'Only got 0 out of 7 bytes'

pm100, 2024-05-03:
On windows.

apio (yosys, nextpnr, icepack, iceprog) builds and uploads the code  without complaint

I have tried both logic analyzer and io and get the same result

`ValueError: Only got 0 out of 7 bytes.`

fpga is alchrity cu - 100mhz ice 40 board

I know its the correct , working com port because if I load some code to the board  that uses the port and  open a serial terminal connection to the board :-
- works fine (my code that is)
- manta complains that it cannot open the port because its in use
 
(Note that the code I am using with manta is not using the serial port, I used different code to verify that the serial port works correctly)

config file

```
cores:
  my_io_core:
    type: io

    inputs:
      val0: 4

  my_logic_analyzer:
    type: logic_analyzer
    sample_depth: 4096
    trigger_location: 1000

    probes:
      value: 4
      sw0: 1
    triggers:
      - sw0 RISING

uart:
  port: "COM6"
  baudrate: 115200
  clock_freq: 100_000_000

```

reading val0 gives

```
>>> manta.my_io_core.get_probe("val0")
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "C:\Users\paulm\AppData\Local\Programs\Python\Python312\Lib\site-packages\manta\io_core.py", line 233, in get_probe
    datas = self._interface.read(probe["addrs"])
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\paulm\AppData\Local\Programs\Python\Python312\Lib\site-packages\manta\uart\__init__.py", line 174, in read
    raise ValueError(
ValueError: Only got 0 out of 7 bytes.
>>>
```
same fail stack with logic analyzer

```
PS C:\work\alchrity_cu\io_tests> manta capture manta.yaml my_logic_analyzer capture.vcd top.v
Traceback (most recent call last):
  File "<frozen runpy>", line 198, in _run_module_as_main
  File "<frozen runpy>", line 88, in _run_code
  File "C:\Users\paulm\AppData\Local\Programs\Python\Python312\Scripts\manta.exe\__main__.py", line 7, in <module>
  File "C:\Users\paulm\AppData\Local\Programs\Python\Python312\Lib\site-packages\manta\cli.py", line 151, in main
    capture(argv[2], argv[3], argv[4:])
  File "C:\Users\paulm\AppData\Local\Programs\Python\Python312\Lib\site-packages\manta\cli.py", line 95, in capture
    cap = la.capture()
          ^^^^^^^^^^^^
  File "C:\Users\paulm\AppData\Local\Programs\Python\Python312\Lib\site-packages\manta\logic_analyzer\__init__.py", line 197, in capture
    state = self._fsm.registers.get_probe("state")
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\paulm\AppData\Local\Programs\Python\Python312\Lib\site-packages\manta\io_core.py", line 233, in get_probe
    datas = self._interface.read(probe["addrs"])
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\paulm\AppData\Local\Programs\Python\Python312\Lib\site-packages\manta\uart\__init__.py", line 174, in read
    raise ValueError(
ValueError: Only got 0 out of 7 bytes.
```

fischermoseley, 2024-05-12:
I just whipped up a quick Icestick project with your config file and unfortunately I'm not able to reproduce the issue. 

I did have a peek at your source at [pm100/alchrity_cu](https://github.com/pm100/alchrity_cu) and I noticed that you're passing the top-level reset directly to `manta`, but according to the Alchrity Cu's schematic that reset appears to be active-low. Manta's expecting an active-high reset, maybe it'd be worth inverting it (or setting Manta's `rst` port to zero)? 

Are you able to port any of the designs in `examples/icestick` to the Alchrity Cu? 

This also might be a duplicate of #18, I'll keep poking around over there.

pm100, 2024-05-14:
That fixed it thank you

closed 2024-05-14 by fischermoseley