fischermoseley/manta #31: Optimize Message Format/Protocol

fischermoseley, 2024-12-06:
## Summary

Manta's current packet messaging format (described [here](https://fischermoseley.github.io/manta/latest/architecture/#message-format)) was designed to be extremely simple and human-readable in order to aid debugging. However, as the interface has been battle-tested and become more stable, debugging hasn't been as necessary, and the lingering performance impact of its design is growing heavier. This is beginning to impair the usability of the tool, and I think it's time to redesign both the message format and the protocol that sends them.

## Problems:
- Encoding. All messages are currently ASCII-encoded, which isn't terribly necessary anymore. A more dense encoding scheme (like [COBS](https://en.wikipedia.org/wiki/Consistent_Overhead_Byte_Stuffing)) would greatly improve bandwidth.

- Reliable packet delivery. This is an assumption baked into the current messaging protocol, which was a little risky when Manta only supported UART, but must be revoked now that UDP is an option. So far I've only been using Manta over UART (which is inherently point-to-point, and only travels short distances) and uncongested networks (like a direct connection to a USB-Ethernet adapter on a host machine). Manta should support UDP traffic over the open (unreliable) internet, and adding a sequence number to each packet should let Manta guarantee that. Adding a checksum for UART would also be good - one isn't necessary for Ethernet as UDP has its own checksum (in addition to the Ethernet FCS).

- Multi-word reads and writes. Currently only one 16-bit word is operated on at a time, which incurs a bunch of overhead when transferring large amounts of data. Support for some kind of bulk read/write would greatly improve performance, primarily with the `MemoryCore` and `LogicAnalyzerCore`.

- Stuffing into UDP packets. Only a single read/write transaction is being put into each UDP packet, which is horribly inefficient! This is being done to prevent the FPGA from having to decide how many messages to put into a UDP packet. Currently the FPGA only sends fixed-length read responses across the wire, so fixing the current interface would only require a counter. Unfortunately this would require delimiters within the UDP packet itself, which I'd like to avoid.

I've sketched out some ideas for this before, but I'll need to properly type them up before starting any design work.

jeremyherbert, 2025-01-01:
Hi,

I have recently discovered manta and it is extremely interesting! Thank you for working on this and releasing it.

One thing I have sort of discovered occurred when playing with the `uart_memory_core.py` amaranth example. Basically, I had more than one FTDI device connected to the PC, and manta detected the incorrect one (which is not really a problem - it just happens). However, there were no errors about being unable to deliver the packets or execute the write. Indeed, I connected an FTDI adapter that goes nowhere and tried to run the code against it and there were no errors printed, but the TX LED blinked. Given that this is a debugging/testing tool, I suppose I expected it to be very verbose about any errors or anomalies in communication. 

I can imagine that there being no response to a packet is for performance reasons, but perhaps there could be an initial link check handshake at the start or something like that.

I believe this is sort of covered in your reliable delivery point above, but I just wanted to add my own experience in case you haven't considered this scenario yet.

fischermoseley, 2025-01-01:
This is a good catch - thank you! In Manta's current communication scheme the FPGA will send a response to a read request, but not to a write request. And since the example you're running only issues write requests, the host machine doesn't expect any bytes from the FPGA, so it'll happily go on sending bytes into the void.

You're right that this should be addressed by the reliable delivery point above - in the new scheme I've been playing with (and still need to type out) both read and write requests should produce a response from the FPGA.