fischermoseley/manta #38: Manta is not sending data via UART

nsokolovic7, 2025-04-03:
Recently we have been working on implementing Manta's logic analyzer on the AX7203 FPGA board, but we can't seem to get the FPGA to send any data via UART. Both RX and TX stay at logic high regardless of the trigger or trigger mode that we select. We have tried using the RISING trigger and we have verified that the probe assigned as the trigger does change value, but still we aren't getting any data from UART.

fischermoseley, 2025-04-04:
Do you have any code you can share? Unfortunately I'm not able to do much without an idea of how you're configuring/instantiating/using the tool. First thing that comes to mind is the polarity of your reset - Manta expects an active-high reset, is that what you're providing it?

Let's keep the conversation on the issue in this repo, that way it's easier for other folks using Manta to find in case they run into the same thing.

nsokolovic7, 2025-04-04:
This is our configuration file:
````
cores:
  my_logic_analyzer:
    type: logic_analyzer
    sample_depth: 2048             # Number of samples captured
    trigger_mode: single_shot      # Options: single_shot, incremental, immediate

    probes:
      probe0: 32 #we will use probe0 as a timer
      probe1: 4 #and probe1 for data transfer 

    triggers:
      - probe1 RISING             # We follow the timer

uart:
  port: "COM5"
  baudrate: 115200
  clock_freq: 200000000
````

and here is our top verilog module:
````
`timescale 1ns / 1ps

module led_test 
(  
    input           sys_clk_p,          // system clock positive
    input           sys_clk_n,          // system clock negative
    input           rst_n,               // reset, low active            
    output [3:0]    led,                 // LED, use for control the LED signal on board
    output          uart_txd,            // UART transmit data
    input           uart_rxd             // UART receive data

);            

// Internal signals
wire            sys_clk;  
reg [31:0]      probe0;
reg [3:0]       probe1;

// Differential clock buffer
IBUFDS sys_clk_ibufgds
(
    .O  (sys_clk),
    .I  (sys_clk_p),
    .IB (sys_clk_n)
);

// Cycle counter and LED/Probe control
always @(posedge sys_clk or negedge rst_n)    
begin
    if (~rst_n) begin  
        probe0 <= 32'd0;
        probe1 <= 4'b0000;
    end else begin
        // Timer increment logic
        if (probe0 == 32'd199_999_999) begin
            probe0 <= 32'd0;
            probe1 <= probe1 + 4'd1;
        end else begin
            probe0 <= probe0 + 1'b1;
        end
    end
end

// Output assignments
assign led = probe1;

// Manta Logic Analyzer
manta manta_inst (
    .clk(sys_clk),
    .rst(rst_n),
    .rx(uart_rxd),
    .tx(uart_txd),
    .probe0(probe0),
    .probe1(probe1)
);

endmodule
````

I have tried sending active-high to the reset, but the outcome was the same.

fischermoseley, 2025-04-04:
Thanks for sharing! I agree with @ZipCPU - Manta will use that reset signal synchronously, and if the rest of your logic uses it asynchronously there's likely to be a timing issue there. Especially since Manta has a few combinational paths hanging right off the logic analyzer core's input ports. Is your design able to meet timing?

If so, then the only other thing I can think of is checking the constraints file/board schematic to see if TX/RX have the direction you expect. In the constraints, is `uart_rxd` an input and `uart_txd` an output? And does that match what you see on the schematic?

nsokolovic7, 2025-04-04:
I have modified the code to sync the reset, so far it hasn't changed anything, here is the now modified code:
````
timescale` 1ns / 1ps

module led_test 
(  
    input           sys_clk_p,          // system clock positive
    input           sys_clk_n,          // system clock negative
    input           rst_n,              // reset, low active            
    output [3:0]    led,                // LED, control signal for onboard LEDs
    output          uart_txd,           // UART transmit data
    input           uart_rxd            // UART receive data
);            

// Internal signals
wire            sys_clk;  
reg [31:0]      probe0;
reg [3:0]       probe1;

// Synchronised reset signals
reg [1:0]       rst_sync;
wire            rst;  // active-high synchronized reset

// Differential clock buffer
IBUFDS sys_clk_ibufgds
(
    .O  (sys_clk),
    .I  (sys_clk_p),
    .IB (sys_clk_n)
);

// Reset synchronizer
always @(posedge sys_clk or negedge rst_n) begin
    if (~rst_n)
        rst_sync <= 2'b00;
    else
        rst_sync <= {rst_sync[0], 1'b1};
end

assign rst = ~rst_sync[1]; // synchronized active-high reset

// Cycle counter and LED/Probe control
always @(posedge sys_clk or posedge rst) begin
    if (rst) begin  
        probe0 <= 32'd0;
        probe1 <= 4'b0000;
    end else begin
        if (probe0 == 32'd199_999_999) begin
            probe0 <= 32'd0;
            probe1 <= probe1 + 4'd1;
        end else begin
            probe0 <= probe0 + 1'b1;
        end
    end
end

// Output assignments
assign led = probe1;

// Manta Logic Analyzer
manta manta_inst (
    .clk(sys_clk),
    .rst(rst),          // synchronized active-high reset
    .rx(uart_rxd),
    .tx(uart_txd),
    .probe0(probe0),
    .probe1(probe1)
);

endmodule
````

I have checked the datasheet again, RX is an input for the FPGA and TX is an output. I also checked and the correct pins are assigned to the RX and TX signals.

fischermoseley, 2025-04-04:
Does this meet timing? Perhaps try dropping the resets from your sensitivity lists in your always blocks.

fischermoseley, 2026-11-15:
Closing due to inactivity.

closed 2026-11-15 by fischermoseley