fischermoseley/manta #43: The issue of insufficient RAM to receive data via UART

Trtn1505, 2025-07-08:
Hello everyone, I am currently working on a research project exploring how to use MANTA to capture signals from digital system lab exercises at my university, and to demonstrate the effectiveness of MANTA. While working on this, I noticed that when I try to capture signals continuously using features like Immediate or Single-Shot mode, the memory fills up in less than one second and data reading begins immediately. As for the remaining capture mode, Incremental, it’s quite unstable—it sometimes works effectively and sometimes doesn’t. I’m currently trying to find a solution and would love to exchange ideas to quickly find the most effective approach. On another note, the exercises I’m working on are related to digital system configurations using the Altera DE2-115 FPGA board, which makes memory configuration more challenging

Trtn1505, 2025-07-08:
các file của tôi có 
-------------
module BAI4B(
    output reg [3:0] Q,
    input wire D,
    input wire clk_4,
    input wire rst_4,
    input wire P
);
    always @(posedge clk_4)
        if (rst_4)
            Q <= 4'b0;
        else
            Q <= {Q[2:0],D};
endmodule
-------------
module BAI4B_test(
    output [0:0] LEDR,
    output [3:0] LEDG,
    input [1:0] SW,
    input [1:0] KEY,
    input wire CLOCK_50,
    inout [1:0]GPIO
);
    assign LEDR[0] = SW[0];
    BAI4B DUT(
        .Q(LEDG[3:0]),
        .clk_4(KEY[1]),
        .rst_4(KEY[0]),
        .D(SW[0]),
        .P(SW[1])
    );
    manta manta_inst(
    .clk(CLOCK_50),
    .rst(0),
    .rx(GPIO[0]),
    .tx(GPIO[1]),
    .Q(LEDG[3:0]),
    .D(SW[0]),
    .clk_4(KEY[1]),
    .rst_4(KEY[0]),
    .P(SW[1])
    );
endmodule
-----------------------
uart:
  port: "COM10"
  baudrate: 9600
  clock_freq: 50000000
cores:
  my_logic_analyzer:
    type: logic_analyzer
    sample_depth: 128
    trigger_mode: single_shot
    probes:
      Q: 4
      D: 1
      clk_4: 1
      rst_4: 1
      P: 1
    triggers:
      - P RISING

fischermoseley, 2025-08-19:
Apologies for the delayed response! This seems like a duplicate (or very much related to) #44. A lot of the same comments from my first reply there (code formatting, build logs/timing reports) also apply here. What drives `clk_4`? Trying to clock off the edge of a register (or sample a clock on another clock's edge) isn't a well-defined timing problem, which makes me think you're trying to do something other than what your RTL is expressing?

fischermoseley, 2026-11-15:
Closing due to inactivity.

closed 2026-11-15 by fischermoseley